librespacefoundation/upsat-obc-hardware #6: Re-design OBC PCB

ppapadeas, 2016-02-10:
On the camera PCB as approved by post-schism meeting. (see upsat-org/issues/20 )

ppapadeas, 2016-02-10:
PCB dimensional specs can be found here:
https://owncloud.librespacefoundation.org/public.php?service=files&t=1ddd58356de674423263c7dc948a11fb

ppapadeas, 2016-03-12:
Outer edge dimensional specs could follow this: https://owncloud.librespacefoundation.org/public.php?service=files&t=04128b0a20f9bfa0fad106c8898eb67f

zisi, 2016-04-04:
Pending review.

ykoveos, 2016-04-04:
Looks ok!
Just a general comment, in ADCS and OBC we have 10u capacitors after mosfet switches. We must check if voltage drop at the main 3.3V rail is ok when switching on the mosfet .
 In any case when switching on, we can have weak pull down and then active down, or have a few short pulses before permanent LOW, to emulate a soft start.

closed 2016-04-04 by ykoveos

zisi, 2016-04-04:
Thank you!! :rocket: