librespacefoundation/upsat-obc-hardware #13: OBC RTC with internal battery

nchronas, 2016-03-11:
http://www.st.com/web/en/resource/technical/document/datasheet/CD00189496.pdf

ykoveos, 2016-03-11:
Perhaps this D-TCXO is better? (temperature range, accuracy).
NO battery, but with a 100uF@3.3V gives ~(1.6-5.5)minutes  (should be enough for a reset(?)
http://gr.mouser.com/ProductDetail/AVX/KR3225Y32768EAW30TAA/?qs=sGAEpiMZZMt8oz%2fHeiymAFQDegnDO61jN7wUajtg4vg%3d

nchronas, 2016-03-11:
If there isnt any problem with the QB50 specs (due to the interal battery), the ST RTC solves the issue of needing the GPS for time. We can preprogram the time before launch and let the RTC work without any external power.

ykoveos, 2016-03-11:
Ok for general housekeeping but QB50-SYS-1.4.4  needs 0.5s accuracy during science operation!! They suggest GPS or uplink correction to achieve this. The best TCXO i found is at 0.2ppm (0.5s @1month)

zisi, 2016-03-25:
For timekeeping we choose the RTC of stm32f405 with battery and external crystal (http://gr.mouser.com/ProductDetail/ABRACON/ABS07-32768KHZ-7-T/?qs=2nyfZ6BV3oibNEFok2qBLA%3D%3D). We choose lithium coin cr1632 battery (http://data.energizer.com/PDFs/cr1632.pdf), according to this parameters:
Battery Capacity: 130mAh
Hrs 1% Time to Fail at +110°C: 1350
°C Operating Temp: 25
Ibat(max): 11000nA, Ibat(typ): 1680nA (stm32f405 in VBAT mode, VBAT=3.3V,Backup SRAM ON, low-speed oscillator and RTC ON, http://www.st.com/web/en/resource/technical/document/datasheet/DM00037051.pdf page 90). The battery life is calculated, https://www.maximintegrated.com/en/design/tools/calculators/product-design/battery.cfm , ~490days(for Ibat(max)), ~3000days(for Ibat(typ)).
The above components have operating temperature -30 - 60°C in comparison with http://www.st.com/web/en/resource/technical/document/datasheet/CD00189496.pdf .
For better timekeeping accuracy (±5ppm (±0.432 Second/Day) from -45°C to +85°C), we find https://datasheets.maximintegrated.com/en/ds/DS3231M.pdf .

ykoveos, 2016-03-25:
Ok, but why not use this as an external crystal? 
http://eu.mouser.com/ProductDetail/AVX/KT3225T32768EAW30TAA/?qs=sGAEpiMZZMt8oz%2fHeiymAFQDegnDO61jjCaBSL6WX7w%3d
you will get the same accuracy with DS3231 or better if you calibrate (see AN3371, but not necessary), and have DS3231 only as a backup?. But you will have to connect Vbat  also to DS3231, otherwise i don't see why to have it. 
No problem for the battery (490days if VDD is not present !!)  just don't add capacitor in parallel or diode for reverse charging, stm has an internal switch for that

zisi, 2016-03-29:
This external crystal http://eu.mouser.com/ProductDetail/AVX/KT3225T32768EAW30TAA/?qs=sGAEpiMZZMt8oz%2fHeiymAFQDegnDO61jjCaBSL6WX7w%3d needs to connect it direct with Vbat (while it has low power consumption) in compare to DS3231 (that it has power control).

We have not consider to use DS3231 as a backup RTC but as alternative choice for RTC. Do you think that is necessary to have a backup RTC?

ykoveos, 2016-03-29:
No I don't think backup RTC is necessary,
but it may be simpler to use KT3225T with a low leakage diode (e.g BAV170T) + R.?
 Then, the absolute max charge delivered is <1mAh/year, and add an R( e.g 10kΩ@5.5uA ->55mV) to be UL safe. It will work with coin voltage down to <2V

e.g see
[vbat_1.pdf](https://github.com/librespacefoundation/upsat-obc-hardware/files/194120/vbat_1.pdf)

and  p.22
http://www.rayovac.com/~/media/Rayovac/Files/Product%20Guides/42691_Lithium%20Application%20Notes%20and%20Product%20Data%20Sheets.ashx

ykoveos, 2016-03-30:
Also ok for DS3231, but you will have to implement the I2C reset in software, in case of a jam, as we discussed before (up to 9? pulses in SCL until SDA is released). It is stated that it is supported, so whatever suits you

closed 2016-04-05 by ppapadeas